odriverobotics/ODriveHardware #5: Trinamic Drivers and new encoder?

npgoalkeeper, 2018-10-01:
Trinamic has a series of 3 phase stepper motor drivers, and when combined with an ATSAMD21G18, can provide high speed closed loop control (almost said stepping). As well as this, Broadcomm has released a 17 bit magnetic encoder that should be more accurate than a standard external quadrature encoder.

madcowswe, 2018-10-01:
Thanks for the information. Can you give me more information on the broadcom chip?

npgoalkeeper, 2018-10-02:
Heres the link to it: https://www.broadcom.com/products/motion-control-encoders/magnetic-encoders/aeat-8800-q24

It works by putting a megnet on the rotor then putting the IC directly below it. It has an extreme 17 bit resolution at a cheap price. It uses an ABI and UVW interface only, however. But that shouldn't be an issue with the ATSAMD21G18A-AUT

RGD2, 2018-10-02:
My take from scanning the datasheet: Pretty cool. 14500 RPM max in ABI mode
(1 MHz max pulse rate) and “near zero” latency, but actually “only” 12 bit
resolution maximum in ABI. Can select lower to allow for higher max RPM
though. Very nice, potentially cleaner/cheaper/more robust than an optical
encoder.

The “high res” stuff (>12 bits) only works for it in absolute mode, with
correspondingly longer latency as you must read it out via a SPI access.
4096 counts per revolution is pretty high already though.

Nice.

madcowswe, 2018-10-03:
Yes, it only has 17bits of resolution when averaging over a period that is too long to be useful in servo motor control.

In servo control, the most important factor is being able to tell the velocity accurately, and therefore the most important metric is the RMS noise, and sufficient bandwidth. In "Fast Mode", this encoder has sufficient bandwidth. However, the noise is specified as "The relative accuracy of the edges to each
other at a CPR setting of 256 is within ± 10%". Because they take CPR to mean "Cycles per revolution", this is equivalent to a noise of 1/2560 of a turn = about 0.1 deg RMS.

This is really bad, worse than all other encoders in the [ODrive encoder guide](https://docs.google.com/spreadsheets/d/1OBDwYrBb5zUPZLrhL98ezZbg94tUsZcdTuwiVNgVqpU). Also the Mechaduino guys did some tests of the same chip (but in a different package, AEAT-6600-T16) [here](https://hackaday.io/project/11224-mechaduino/log/37308-first-prototype), and concluded it wasn't very good.

As for the trinamic chip, I don't see any features that we don't have (or could easily get) on our firmware based solution. Why sacrifice the flexibility?

Nevertheless, thanks for the heads up.

closed 2018-10-03 by madcowswe